sparkfun/MicroMod_Asset_Tracker #7: ESP32 SARA-R510S |  Cannot initialize SARA with Examples

alex-dsbz, 2022-07-13:
Hi. 

I'm having issues with running any examples that include the initialisation of the SARA board. All of the other examples work with no issue.

Let's take Example12 - Identification. 
When uploading it to the MicroMod ESP32, this is what I get from the serial monitor of Arduino

````
Asset Tracker Example

Initializing the SARA-R5...
...this may take ~25 seconds if the SARA is off.
...it may take ~5 seconds if it just turned on.

Press any key to continue...
init: Begin module init.
sendCommandWithResponse: Command: E0
init: Module failed echo test.
init: Begin module init.
init: Attempting autobaud connection to module.
sendCommandWithResponse: Command: +IPR=9600
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=9600
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=9600
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=9600
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=9600
sendCommandWithResponse: Command: 
sendCommandWithResponse: Command: +IPR=9600
sendCommandWithResponse: Command: 
init: Begin module init.
init: Power cycling module.
powerOff: complete
powerOn: complete

````
This sequence loops a couple time before its asking to manually turn on the SARA.

However the SARA IS turned on, and is being turned on and off by the initialisation script.

I've posted on the Sparkfun forum, but did not get any assistance so far.

I would love to be pointed in the right direction as I've exhausted all I could think off...

PaulZC, 2022-07-13:
Hi @alex-dsbz ,

From the title of the issue, I guess you are using the ESP32 MicroMod Processor Board?
Which version of the Espressif ESP32 board package do you have installed? (Probably v2.0.3 or 2.0.4?)
Which LEDs are illuminated?
Have you made any changes to the jumper links on the board?

Many thanks,
Paul

alex-dsbz, 2022-07-13:
Hi Paul,

Thanks for reaching back, I really appreciate it!

So I'm using the AWS IoT ExpressLink SARA-R5 Starter Kit  https://www.sparkfun.com/products/18450  

All I have done is swap the Asset Tracker Update Tool MicroMod with the ESP32 and plug the carrier board by USB-C to my macbook. 

I have uploaded the following example https://github.com/sparkfun/MicroMod_Asset_Tracker/tree/main/Examples/AssetTracker_Example12_Identification

In terms of Boards package, I have two installed:
- esp32 by Espressif Systems version 1.0.6 (that's the latest according to the Arduino IDE)
- Sparkfun Esp32 Boards version 1.0.1

[image]

In terms of LED:
- VIN is orange
- 3C3 is red
- On is white

I hope that answers your questions :)

Alex

PaulZC, 2022-07-13:
Hi Alex (@alex-dsbz ),

Many thanks for the detailed reply.

The SARA-R5 on the AWS IoT ExpressLink version of the Asset Tracker was programmed with special firmware. From the product page:

[image]

I must confess, I've never had one of those boards to test.

I designed the original version of the board and wrote the original Arduino Library and Examples. But I did all of that with boards running the standard SARA-R5 firmware - which responds to 'regular' AT commands.

I suspect the AWS version of the module is ignoring or not responding to the initial **E0** 'ping'. But, I've got no way to confirm that just now...

I'm really not sure how best to proceed.

I can ask my colleagues to send over one of the AWS boards so I can replicate your set-up. But, the end result _could_ be that it's just not possible to use the Arduino examples with that board...

Apologies,
Paul

alex-dsbz, 2022-07-13:
Oh I see, that does make sense!
At least now I know, there is nothing wrong with the board :) 

Thanks a lot for your help!

closed 2022-07-13 by alex-dsbz

PaulZC, 2022-07-23:
Hi Alex,

Just adding a link to the forum issue, just in case anyone else has the same issue:

https://forum.sparkfun.com/viewtopic.php?p=234780#p234780

Very best wishes,
Paul