sparkfun/MicroMod_Asset_Tracker #8: MicroMod Asset Tracker Carrier board with Samd51, Arduino IDE giving Missing FBQN error

jwalway, 2022-10-21:
Hello,

I have a **SparkFun MicroMod Asset Tracker Carrier** board that has a **Samd51 processor** board plugged into it.
I'm using the **Arduino IDE 2.0** in an attempt to code the board. I'm on _Windows 10_ and the board is plugged into my usb com port. Right now I'm just experimenting. I'm relatively new to this kind of programming. I downloaded libraries for both boards. When I attempt to compile a short sketch/script I get the following error: **Missing FBQN** (Fully Qualified Board Name). I can't get the Arduino IDE to recognize the Asset Tracker board, nor do I see it listed among their boards. I also don't see the samd51. I do see samd21.

I'm trying the first example here:
[https://learn.sparkfun.com/tutorials/mi ... o-examples](https://learn.sparkfun.com/tutorials/micromod-asset-tracker-carrier-board-hookup-guide/arduino-examples)

Can anyone help me with this?

Thanks!
...John

jwalway, 2022-10-21:
> 

Thanks PaulZC

I'm following that guide, but am running into an issue here:
[Samd51 Hook up guide](https://learn.sparkfun.com/tutorials/micromod-samd51-processor-board-hookup-guide/setting-up-the-arduino-ide)

Where it says:
_First, you'll need to install a variety of tools, including [low-level ARM Cortex libraries](http://www.arm.com/products/processors/cortex-m/cortex-microcontroller-software-interface-standard.php) full of generic code, [arm-gcc](https://launchpad.net/gcc-arm-embedded) to compile your code, and [bossa](http://www.shumatech.com/web/products/bossa) to upload over the bootloader. These tools come packaged along with Arduino's SAMD board definitions for the Arduino Zero._

For the arm-gcc they link to I don't see exactly what to download.  I'm here on Windows 10.   Put another way, I'm not sure what to download.

PaulZC, 2022-10-22:
Hi John,

Let's see if we can get you going. I know there are some issues in IDE 2.0 that will be important here and which will be corrected in the next release. But let's see if standard 2.0 works...

Steps:

* Install IDE 2.0
* Open File \ Preferences
* Click on the icon to the right of "Additional board manager URLs"
* Copy and paste in the following:

```
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/main/IDE_Board_Manager/package_sparkfun_index.json
```

* Click OK
* Open Tool \ Board \ Boards Manager  or click the icon second from top at the left of the screen ("Boards Manager")
* In the search box type "SAMD"
* Install Arduino SAMD Boards (32-bits ARM Cortex-M0+) **Version 1.8.13**
* Scroll down the list, install SparkFun SAMD Boards Version 1.8.7
* Click on the icon on the left of the screen to close the board manager
* Click on the books icon below it to open the Library Manager  or open Tools \ Manage Libraries
* In the search box type "sparkfun sara" - that's sparkfun space sara
  * You can't use underscores at the moment. That will be fixed in the next release
* Install the SparkFun u-blox SARA-R5 Arduino Library Version 1.1.3
* In the search box, replace sara with gnss
* Scroll down and install the SparkFun u-blox GNSS Arduino Library Version 2.2.17
* Go to this page:
  * https://github.com/sparkfun/MicroMod_Asset_Tracker
* Click on the green Code icon
* Select Download Zip
* Go to your downloads folder and unzip the MicroMod_Asset_Tracker-main.zip
* Go into the Zip file Examples folder
* Rick-click and Copy the AssetTracker_Example1_Blink folder
* Go to your Documents \ Arduino folder and paste it there
* In the IDE, click the books Library Manager icon on the left to close the Library Manager
* Click on the folder icon above it and open your "Sketchbook"
* Highlight the AssetTracker_Example1_Blink
* Click on the three dots to the right
* Select "Open Sketch In New Window"
* The example code will pop up in a new window
* Click on the AssetTracker_Example1_Blink.ino tab to see the main example code
* Install the SAMD51 processor and connect the Asset Tracker via USB as per the hook-up guide
* Click on the Select Board drop-down list at the top left of the window, below Help
* Select the SparkFun SAMD51 MicroMod
* Click on the -> Upload icon at the top left of the window, below Edit
* The code should compile and upload
* The blue LED on the processor board should start to flash

Your screen should look like this when you're done:

[image]

If you need to select the board and port manually, use:

* Tools \ Board \ SparkFun SAMD Boards \ SparkFun SAMD51 MicroMod
* Tools \ Port
  * Select the COM port which has SparkFun SAMD51 MicroMod named next to it

You shouldn't need to install any additional USB drivers - fingers crossed!

That will hopefully get you started! If you see any issues, please report back here. Please cut and paste in screenshots to help.

Best wishes,
Paul

jwalway, 2022-10-22:
Thanks Paul!

That was very thorough!

I had previously done every step you mentioned, except that I didn't have the _SparkFun u-blox GNSS Arduino Library_ installed.   So I installed that.

**There were a couple of other differences:**
1) I can't find the _SparkFun SAMD Boards version1.8.7_... previously I had installed the _1.8.3 version_.

2) I had installed the following url in preferences:

https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json

Yours is slightly different.  So, I added yours as well (the master vs main in url string)

https://raw.githubusercontent.com/sparkfun/Arduino_Boards/main/IDE_Board_Manager/package_sparkfun_index.json

**Everything went smoothly until I came to the final step, which is where my main trouble was.**

I can't see the board to select for the COM port.    I see samd21 versions, but **no samd51**.

I've uploaded a screenshot...

[image]

Perhaps this is the one problem I have.

PaulZC, 2022-10-22:
Hi John (@jwalway ),

Please delete the “master” entry from the additional URLs. Close and re-open the IDE. Open the Library Manager. Search for SAMD. Do you see an option to update the SparkFun SAMD boards to 1.8.7?

The issue - I think - is that the IDE is using the old ‘master’ link preferentially. That hasn’t been updated in some time and probably doesn’t include the SAMD51 PB.

Best,
Paul

jwalway, 2022-10-22:
Thanks a million, Paul!   That fixed it.  It's doing the one second led on/off cycle!    It's a beautiful thing. :-)

Regards,
...John

PaulZC, 2022-10-22:
Happy days!

The GNSS library was probably a red herring. I just had a vague memory that some of the later examples needed it. The SARA-R5 has a GNSS built-in, but you can’t access it directly. You need to do it through the SARA’s AT command interface.

Anyway… I’m going to close this, but will take an action to update any existing master references in our guides. GitHub and we both updated the branch name to main some time ago.

Please open a fresh issue if you find any more gremlins…

Best wishes,
Paul

closed 2022-10-22 by PaulZC