sparkfun/MicroMod_Asset_Tracker #9: Serial monitor doesn't seem to work on my setup

SelimOzel, 2022-12-29:
Hi There. I got the Micromod Asset tracker and the Sparkfun STM32 processor board. I can run the blink example and upload the code. I already did the software and hardware setup for setting up SARA library and board definitions. 

However, I can't seem to get anything out on the serial monitor. I'm wondering if the default serial initialization in the examples has a mismatch or not.

I can provide more details if it'll help.

Best,
Selim

PaulZC, 2022-12-29:
Hi Selim (@SelimOzel ),

There is something strange going on... Possibly to do with auto-bauding? Let me explain:

I started by opening Asset Tracker Example12 (Identification) in the Arduino IDE 1.8.19. I tried to upload the code to the Asset Tracker and STM32 MicroMod processor. After remembering to hold the Boot button while pressing and releasing the Reset button, I got the code to upload OK. Like you, I could not see any serial traffic in the Serial Monitor.

I inserted a ```delay(2000);``` at the start of ```setup()```. That allowed me to see the initial "Asset Tracker Example" messages. When I pressed a key to continue, I eventually got a "Unable to communicate with the SARA" message.

I uncommented the ```assetTracker.enableDebugging(SERIAL_PORT);``` to see the debug messages. I reset the board and pressed a key again. I could see the library code attempting to communicate with the SARA, but it was not receiving any replies. Eventually it timed out.

I removed the STM32 MM processor and inserted the Firmware Update Tool instead. I installed [m-center](https://www.u-blox.com/en/product/m-center). I opened the CH340 COM port at **9600 Baud** (no handshaking) and was able connect to the SARA and see its serial number etc..

I then re-installed the STM32 processor and re-ran the example code. This time it worked:

[image]

So, I'm not sure what is going on... I suspect the SARA had accidentally been set to a different Baud rate and that opening m-center somehow reset it?

Please check you have:
* U(S)ART support set to "Enabled (no generic 'Serial')"
* USB support set to "CDC (generic 'Serial' supercede U(S)ART"

[image]

There is more useful information in the [STM32 MM PB Hook-Up Guide](https://learn.sparkfun.com/tutorials/micromod-stm32-processor-hookup-guide#software-setup-and-programming) - but I guess you've already seen that.

I hope this helps. Please let me know if this lets you communicate with the SARA.

Best wishes,
Paul

SelimOzel, 2022-12-30:
Hey Paul,

Thank for taking the time to look into this! I think your suggestion worked. I'll give more details though, maybe it can help others.

I felt something was off with my unit when I went through the Hook-up Guide for the first time. Examples with serial monitor only worked if I added a delay, just like you did, to the beginning of the loop. I also had to remove the code from setup and add it under loop with a "run once" flag. So, I ended up adding a pseudo-setup routine in my code. This seemed to work and I could go through examples 12-13 with my custom code.

At some point this setup broke down. I stopped seeing any serial output on my monitor. Just this morning, I followed what you did. I removed STM32 board and attached the update board, I opened a port and set it to 9600 Baud on m-center. Then, I removed update board and reinstalled STM32 board. After I uploaded the custom code, I could see serial output this time!

Apparently uBlox was set to a wrong Baud rate internally and this was breaking the serial communication that I use to debug over USB as well.

Best,
Selim

PaulZC, 2022-12-30:
Thanks for the update Selim - I'm glad that's working for you,

I don't understand what's going on. The STM32 is a little bit tricky - with its USART and CDC configuration. Maybe some rogue traffic is output on the TX1/RX1 pins during the DFU upload - which causes the SARA to auto-baud to the wrong speed? It's just a guess though... (Once the SARA has chosen an auto-baud speed, you can only change it by: resetting the chip, or by sending the correct AT commands at auto-baud speed.)

I'm going to close this, but I will pin it so others can find it.

Thanks again for the diagnostics,
Paul

closed 2022-12-30 by PaulZC