sparkfun/MicroMod_Asset_Tracker #11: Reducing power consumption of Micromod Asset Tracker

jeronimoagullo, 2024-06-03:
Hi!

I would like to know the lowest current consumption of the Micromod Asset Tracker board (without the CPU consumption).

I have removed the CPU board and I have measured around 16mA feeding the board with 3.3V and with the modem off. I noticed that the IMU was enabled, but putting "ICM20948_EN" pin of its voltage regulator to GND, the current consumption reduces a couple to milliamps to 13-14mA.

I am wondering which peripherals are consuming these 13-14mA and how could I eliminate this remaining power consumption by configuring the CPU pins properly. The idea es to reduce the power consumption during sleep state (I am using the nrf52840 board which consumes less than 1mA in idle state)

PaulZC, 2024-06-04:
Hi Jerónimo (@jeronimoagullo ),

The Asset Tracker wasn't really designed with low current in mind. You will need to work through the schematic, tracing where those mA are going.

Some suggestions:

* Powering the board from 3.3V may not be the best choice for measuring current draw. You may be leaking current back through the 3.3V regulator. What is the draw if you provide power from VIN or V_BATT?
* Are all the LEDs off? Have you disabled the VIN and 3.3V LEDs by opening the jumper links?
* Do you have an active GNSS antenna connected? The regulator U3 _should_ be off when the SARA is powered off, but do check. Check Q8 and Q9 too.
* What state is the Qwiic I2C bus in? Are you leaking current through the 2.2K I2C pull-ups? Likewise for the level-shifted SARA I2C bus.
* Do you have an SD card inserted? The SD card will be powered on by default. Pull MICROSD_PWR high (3.3V) to disable it.
* Check SPI_SDI. Are you drawing current through the R22 2.2K pull-up? Likewise for SARA_INT and R30.
* Is USB connected?

I'm going to close this, but please feel free to ask here for more help if you need it.

Best wishes,
Paul

closed 2024-06-04 by PaulZC

jeronimoagullo, 2024-06-04:
Hi @PaulZC 

thank you very much for your response. I checked out some of your recommendations but no all of them. Just as a comment, after openning this issue I was checking the level shifters (TI SN74AVC4T774) and it seems that when the inputs are floating, there is a large current leakage, so I suppose it is the main reason. FYI: https://e2e.ti.com/support/logic-group/logic/f/logic-forum/945807/sn74avc4t774-sn74avc4t774rsvr-consuming-high-current?tisearch=e2e-sitesearch&keymatch=SN74AVC4T774%25252525252520consumption#

In any case, the board is awesome, great job :D

Best regards,
Jerónimo